FrameworkComputer/SoftwareFirmwareIssueTracker #105: The FW16's firmware GUI's resolution is significantly less than that of the display.

RokeJulianLockhart, 2025-09-20:
## Device Information

### System Model or SKU

- [x] Framework Laptop 16 (AMD Ryzen™ 7040 Series)

### Firmware Version

1. ~~~sh
   #!/usr/bin/env sh
   sudo dmidecode --string bios-version
   ~~~

1. > 03.05

### DIY Edition information

1. #### Memory

   1. ~~~sh
      #!/usr/bin/env pwsh
      sudo lshw -c memory -json | ConvertFrom-JSON | Format-Table -AutoSize
      ~~~

   1. <blockquote>

      ~~~Markdown
      id       class  claimed description                                                    vendor            physid version date       units        size
      --       -----  ------- -----------                                                    ------            ------ ------- ----       -----        ----
      firmware memory    True BIOS                                                           INSYDE Corp.      0      03.05   11/13/2024 bytes      131072
      cache:0  memory    True L1 cache                                                                         5                         bytes      524288
      cache:1  memory    True L2 cache                                                                         6                         bytes     8388608
      cache:2  memory    True L3 cache                                                                         7                         bytes    16777216
      memory   memory    True System Memory                                                                    15                        bytes 34359738368
      bank:0   memory    True SODIMM Synchronous Unbuffered (Unregistered) 5600 MHz (0.2 ns) A-DATA Technology 0                         bytes 34359738368
      bank:1   memory    True [empty]                                                        Unknown           1
      ~~~

      </blockquote>

   <p></p><details><p></p>

   1. ~~~sh
      #!/usr/bin/env sh
      sudo lshw -c memory -json | yq -P
      ~~~

   1. <blockquote>

      ~~~YAML
      - id: firmware
        class: memory
        claimed: true
        description: BIOS
        vendor: INSYDE Corp.
        physid: "0"
        version: "03.05"
        date: 11/13/2024
        units: bytes
        size: 131072
        capacity: 33554432
        capabilities:
          pci: PCI bus
          upgrade: BIOS EEPROM can be upgraded
          shadowing: BIOS shadowing
          cdboot: Booting from CD-ROM/DVD
          bootselect: Selectable boot path
          int9keyboard: i8042 keyboard controller
          int10video: INT10 CGA/Mono video
          acpi: ACPI
[... 426 lines cut ...]
      ~~~

      </blockquote>

1. #### Linux Kernel Version

   1. ~~~sh
      #!/usr/bin/env sh
      uname -a | jc --uname -p | yq -P
      ~~~

   1. <blockquote>
   
      ~~~YAML
      kernel_name: Linux
      node_name: Beedell.RokeJulianLockhart.laptop.SKQY07
      kernel_release: 6.16.7-200.fc42.x86_64
      operating_system: GNU/Linux
      machine: x86_64
      kernel_version: '#1 SMP PREEMPT_DYNAMIC Thu Sep 11 17:46:54 UTC 2025'
      ~~~

      </blockquote>

## Additional Context

JohnAZoidberg, 2025-09-20:
On the new Framework Desktop we have improved the text and resolution scaling in the upcoming BIOS version 3.03.
We inherit the same behavior that the BIOS vendor provides for all their customers, but we have worked with them to improve it.
We are trying to bring this to the laptop products in the future as well.

JohnAZoidberg, 2025-09-20:
> considering I've 03.05, is 03.03 "upcoming"?

No, I'm referring to the Framework Desktop BIOS 3.03.

We don't have a timeline for Framework 16 AMD Ryzen 7040 yet.